MRX8024/motors-sync #1: TypeError during Sync_Motors resulting in Klipper Crash

YanceyA, 2024-05-05:
This issue has been resolved with the v1.0.0.7 May 5 (50dfe11) commit. I tried both CSV_OPEN_DELAY as 0.1s (default) and 0.5s and both worked with this version. I did not try CSV_OPEN_DELAY at 0.5s with v1.0.0.6 to see if that variable change would fix my issue.

------------------------------------
I got everything setup but everytime i run motors sync I get what looks like a python error during the process. V1.0.0.6

_7 cputime=11.280 memavail=3689296 print_time=86.238 buffer_time=0.000 print_stall=0 extruder: target=0 temp=13.3 pwm=0.000
Writing raw accelerometer data to /tmp/beacon-20240504_213802.csv file
Y-New magnitude: None on -1/16 step move
Internal error on command:"SYNC_MOTORS"
Traceback (most recent call last):
File "/home/yanceya/klipper/klippy/gcode.py", line 298, in _process_commands
handler(gcmd)
File "/home/yanceya/klipper/klippy/gcode.py", line 207, in func
return origfunc(self._get_extended_params(params))
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/yanceya/klipper/klippy/extras/motors_sync.py", line 266, in cmd_RUN_SYNC
self._axes_level()
File "/home/yanceya/klipper/klippy/extras/motors_sync.py", line 155, in axes_level
if m['new_magnitude'] > m['magnitude']:
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: '<' not supported between instances of 'float' and 'NoneType'
Transition to shutdown state: Internal error on command:"SYNC_MOTORS"

I've attached the relevant klippy log:
[klippy_motors_sync_error.txt](https://github.com/MRX8024/motors-sync/files/15211880/klippy_motors_sync_error.txt)

I'm running a monolith AWD and Beacon as the accelerometer sensor.

YanceyA, 2024-05-05:
Closed, with v1.0.0.7 this issue has not reappeared.

closed 2024-05-05 by YanceyA

MRX8024, 2024-05-05:
Hello, thanks for the feedback! Well, the new version could not affect this problem in any way, perhaps rebooting the host or something like that helped you..

-Maxim