MRX8024/motors-sync #5: Internal error on command:"SYNC_MOTORS"

FlorentBroise, 2024-07-02:
[image]

Hi
Got this issue after a full proper installation
Maybe you will be able to guide me ;) I'd love to see this mod to work ;)

Thanks in advance

[klippy (2).log](https://github.com/user-attachments/files/16073981/klippy.2.log)

MRX8024, 2024-07-04:
Hi, thanks for the feedback. I just noticed the same issue from a month ago, due to the lack of notifications. I see you already solved the problem without me and suggested pr. Honestly, I thought that Python can automatically convert float to int like jinja2 does for example :) I think I'll rewrite this function today in a better way, wait for the news!

-Maxim

MRX8024, 2024-07-04:
Well, I fixed the problem. 
Thanks again, if you have no questions or wishes, I think we can close this issue and pr.

-Maxim

FlorentBroise, 2024-07-08:
Thanks a lot! Working like a charm now :)

closed 2024-07-08 by FlorentBroise