MRX8024/motors-sync #7: SYNC_MOTORS with tuner ROTATION_DISTANCE

FlorentBroise, 2024-07-10:
[image]
Hi
As I'm using custom rotation_distance values for XY Z to make perfect dimensionned parts (Not the only one I presume), is there a solution for your mod to work with another value than a round one (40) put with 39.xxx by exemple?

Thanks in advance

MRX8024, 2024-07-15:
Hello, I still dont receive notifications :) I believe that it is incorrect to edit the size of the prints by editing the distance of the pulleys; they are always machined to certain constants. This should be done with a slicer or third-party programs. However, I will add rounding to the nearest integer.

-Maxim

FlorentBroise, 2024-07-19:
Are you aware of another solution to match the dimentionnal accuracy depending of material expension, Nozzle width, and some other factor that can impact accuracies? Scaling is another option but give some other dimentionnal issue...

MRX8024, 2024-07-20:
My prints suit my requirements with normal scaling inside the slicer. If your model geometry (length, width, etc.) changes depending on the width of the nozzle, then when changing the nozzle you will have to use a different `rotation_distance`... However, despite the fact that this does not apply to the current repository, you can describe the problem in more detail , and if I know the answer, I will help you.

closed 2024-12-07 by MRX8024