MRX8024/motors-sync #10: Unknown Config Object "lis2dw"

okeightysix, 2024-09-25:
[image]

Hi after updating to V1.1.0-45-g2ec826dd i start to get this error. 

Any ideas?

MRX8024, 2024-09-25:
Hello, please provide klippy.log

gogsi1, 2024-09-29:
> Hello, please provide klippy.log
![Снимок экрана 2024-09-29 134929](https://github.com/user-attachments/assets/76311efb-3f50-46d4-9533-f5
[klippy.log](https://github.com/user-attachments/files/17178623/klippy.log)
3a7e5b4d21)

MRX8024, 2024-09-29:
Detect your accelerometer before [motors_sync], i.e.:

`[cartographer]`
`...`

`[adxl345]`
`cs_pin = cartographer:PA3`
`spi_bus = spi1`

`[motors_sync]`
`...`

okeightysix, 2024-09-29:
> Detect your accelerometer before `[motors_sync]`, i.e.:
> 
> `[cartographer]` `...`
> 
> `[adxl345]` `cs_pin = cartographer:PA3` `spi_bus = spi1`
> 
> `[motors_sync]` `...`

it worked thank you and sorry for late answer i couldn't be able to provide log since i was away. But after changing the order it worked :)

closed 2024-12-07 by MRX8024