MRX8024/motors-sync #11: SYNC_MOTORS_CALIBRATE not generating graphs

rcopra, 2024-09-26:
I'm assuming something is wrong with my klippy-env but I can't quite pin it down. After running SYNC_MOTORS_CALIBRATE, expected behavior occurs, but once the toolhead stops moving, no graphs are generated or further information is shared in the console.

[image]

I'm not sure how to troubleshoot this issue, so I figured I'd open an issue here.

MRX8024, 2024-09-26:
Hellow, this may be due to a decrease in the priority of the task, check your htop for example for active processes, and let me know if this is the case.

rcopra, 2024-09-26:
This is new for me, apologies for any handholding I'll require to troubleshoot - I ran htop but I wasn't able to identify the correct process to check its priority.

What command should I be looking for in order to identify the correct task? I didn't see anything labeled motor sync during the run.

rcopra, 2024-09-26:
Sorry - accidentally hit close. Reopening.

MRX8024, 2024-09-26:
Open the file `motors_sync.py`, by default it is in `~/motors_sync/motors_sync.py`. If you're currently on the latest version, look for line `639` that says `os.nice(20)`, change that to `os.nice(5)`, re-calibrate and let me know if that helped or if you encountered any other problems.

rcopra, 2024-09-26:
Thanks for the follow-up. No luck unfortunately. I changed the line as you recommended in `motors_sync.py` from `os.nice(20)` to `os.nice(5)` and still was not able to have anything happen after running `SYNC_MOTORS_CALIBRATE`

MRX8024, 2024-09-26:
Please share klippy.log and tell us about your host. Also try changing it to 0 this time :) The process is running, there are no errors, but for some reason it is happening very slowly.

rcopra, 2024-09-27:
Hi, thanka for offering further support :) 

I changed `os.nice(5)` to `os.nice(0)` and ran SYNC_MOTORS_CALIBRATE REPEATS=1 and here's the Klippy log for that. 

It seems after it takes ADXL measurements nothing further occurs. 

[klippy_motors_sync.log](https://github.com/user-attachments/files/17163305/klippy_motors_sync.log)

pisyakot, 2024-09-29:
у меня кстати, подобная проблема. Графиков не дает теперь:) пробовал убирать кусок с try, не помогло :(

MRX8024, 2024-09-29:
I got to my printer and the problem really turned out to be else, I fixed it with an update.
Please check if it works for you.
Thanks.

gogsi1, 2024-09-29:
> I got to my printer and the problem really turned out to be else, I fixed it with an update. Please check if it works for you. Thanks.

It worked for me, before that, I tried all the options that you wrote.

rcopra, 2024-09-29:
Hurrah! It's working again - thanks for pushing such a fast fix. I was about to reflash my CB2

closed 2024-09-29 by rcopra