MRX8024/motors-sync #13: Sync process stops to early

Fragmon, 2024-10-24:
When i use this sync plugin it stops randomly to early. The axles magnitude is high or higher then the start value.
[image]
[image]

I have to re-run the sync process multiple times until i get a low magnitude.

MRX8024, 2024-10-24:
Hi, can I know a little more about your printer? The problem in this case is that the accelerometer displays incorrect information. 

For example in the first screenshot: The X axis determined the initial value at 66K, stepped forward - 69K, which means you need to move back, but the next movement back gave 76K. Because of this, the script terminates prematurely without any checks.

The `retry_tolerance` and `retries` options can help you in this situation, they are designed to protect against incorrect measurements, but for isolated cases, and not when almost every second measurement is incorrect.

“Why does the wrong measurement occur?” - I can't give you a specific answer to this question, you should check your mechanics yourself, perhaps attaching an accelerometer randomally smoothes out vibrations.

Fragmon, 2024-10-24:
Its a trident AWD and i can hear that the axle has a high magnitude but the script dont reduce that and Jump to the other axle.

In the next run he test longer until it reach <1000.

MRX8024, 2024-10-24:
In that case study the functions that I wrote above, and if it helps you, it will be great. Finding out the root cause may not be easy, but it is advisable.

closed 2024-12-07 by MRX8024