MRX8024/motors-sync #14: Getting error when trying to run sync_motors

erlis, 2024-11-02:
I'm getting this error when I'm running SYNC_MOTORS 

[image]

it looks like the steppers array is missing, this is weird, 

```
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 298, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 207, in func
    return origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/motors_sync.py", line 517, in cmd_RUN_SYNC
    init_magnitude = self._measure(axis, True)
  File "/home/pi/klipper/klippy/extras/motors_sync.py", line 309, in _measure
    if buzz: self._buzz(axis)
  File "/home/pi/klipper/klippy/extras/motors_sync.py", line 277, in _buzz
    lookup_sec_stepper = self.motion[axis]['lookuped_steppers'][1]
IndexError: list index out of range
Transition to shutdown state: Internal error on command:"SYNC_MOTORS"
```

erlis, 2024-11-02:
disregard, I just realized I had the two other motors commented out in my config file, and that is the reason it was not found...

closed 2024-11-02 by erlis

MRX8024, 2024-11-02:
This is a good reason to add protection against such cases. 
Thanks, you are welcome.