MRX8024/motors-sync #15: Please add support for kinematics: limited_corexy

Kcdc3d, 2024-11-12:
I would like to use differing accel limits on my x and y axis, which is possible on DangerKlipper by using Kinematics: limeted_corexy

When I tried setting it up, klipper goes to standby red mode because this plugin doesn't currently support this. Is this an easy thing for you to add? Or does having differing accels on each axis disrupt how ytour software is meant to operate? If you can do it, I would greatly appreciate it!

MRX8024, 2024-11-12:
Hi, switch branch with `git checkout limited_corexy`. 
I don’t know how danger klipper works; after a quick look, I didn’t find any conflicts, but they may exist.
Test it, good luck.

Kcdc3d, 2024-11-12:
Amazing, thank you!!

closed 2024-11-12 by Kcdc3d

MRX8024, 2024-11-12:
Well, have you tested it yet, and does it work? then I would have included it in the main branch

Kcdc3d, 2024-11-14:
Please let me know if you need specifics to pass it to master.

Kcdc3d, 2024-11-14:
I tested using PA calibrations from orcaslicer, then a print test using those settings, after syncing motors with the limited_corexy branch.. the print came out as expected, so thankfully your engine/system translates to differing accels as far as I can tell