MRX8024/motors-sync #16: Axes parameter under [motors_sync]

askmewhy25, 2024-11-14:
I'm running Siboor Trident AWD, and since the printer has two additional front motors, it is safer to home the Y axis first before the X axis to make sure it will not hit the front right motor in case the toolhead is too far in front.

I tried checking the Python script at line 73; the default is to home X first, then Y. While in line 334, the homing script should home the first axis defined in the axes parameter under [motors_sync].

I'm not sure if that should be the case. Thank you.

https://github.com/MRX8024/motors-sync/blame/9fec3677ced8806e5a9e8c2116564b884e4de760/motors_sync.py#L73

MRX8024, 2024-11-14:
Hi, I don't quite understand what the problem is. The fact that the front motors interfere with the print head movement is a design problem, which does not apply to global parking. It is customary to park first x, then y. However, if you specify `axes: y,x` in the configuration, then parking will happen first for the Y axis. You can also use the usual `[homming_override]` as other users usually do.

-Maxim

MRX8024, 2024-11-14:
I was wrong about this statement: 
> if you specify `axes: y,x` in the configuration, then parking should will happen first for the Y axis

The command is sent in a different order, but the klipper `cmd_G28()` will still park the X axis first, so you only have 1 option - `[homing_override]`, which is enough

askmewhy25, 2024-11-15:
Hi Maxim, thanks for the suggestion. The solution I found was to include the X and Y axes in the homing override.

closed 2024-11-15 by askmewhy25