MRX8024/motors-sync #18: internal issue

LovesDonuts, 2024-12-08:
after the update, cannot sync at all. first time user, also might be a user error lol

[image]

MRX8024, 2024-12-08:
Hello, I made some quick changes that should fix the problem. But since I don't know the real beacon numbers, it may be optimized in the future. Perhaps I will contact you this week so that you can take some measurements, if you don't mind, since I do not have this sensor myself. 

Thanks.

LovesDonuts, 2024-12-08:
yes that will be no problem. thank you for replying

MRX8024, 2024-12-08:
Well, switch to the branch - `git checkout beacon_batch_test`, restart klipper via `systemctl restart klipper`. Next, just enter into the web terminal `SYNC_MOTORS`, wait a few seconds and send the answer that it will show you. Then you can return to `git checkout main` again. Thanks!

LovesDonuts, 2024-12-08:
4:03 PM
4843.188253877792
4:03 PM
4843.137413935081
4:03 PM
4843.086585117692
4:03 PM
4843.035744518712
4:03 PM
4842.984906138547
4:03 PM
4842.934076727391
4:03 PM
4842.883237034686
4:03 PM
4842.832402592135
4:03 PM
4842.781584556309
4:03 PM
4842.730727726614
4:03 PM
4842.679892753069
4:03 PM
4842.631174623436
4:03 PM
4842.580351243984
4:03 PM
4842.529511676554
4:03 PM
4842.478667358982
4:03 PM
4842.427832354188
4:03 PM
4842.376995505587
4:03 PM
4842.326162782109
4:03 PM
4842.275337027588
4:03 PM
4842.226908969041
4:03 PM
4842.176077308094
4:03 PM
4842.122817137761
4:03 PM
4842.0744030483775
4:03 PM
4842.023567918578
4:03 PM
4841.972438967581
4:03 PM
4841.921593556228
4:03 PM
4841.87077467691
4:03 PM
4841.819926859234
4:03 PM
4841.771508175966
4:03 PM
4841.718271149874
4:03 PM
4841.669840440745
4:03 PM
4841.616579270466
4:03 PM
4841.568164999342
4:03 PM
4841.517328622433
4:03 PM
4841.466493495561
4:03 PM
4841.418085443379
4:03 PM
4841.364830710798
4:03 PM
4841.314000427825
4:03 PM
4841.263171957408
4:03 PM
4841.2123166736665
4:03 PM
4841.161483078092
4:03 PM
4841.110651013814
4:03 PM
4841.059820949598
4:03 PM
4841.013821221529

MRX8024, 2024-12-08:
I did a force push with the correct values. There shouldn't be any problems, but I think we should keep this issue open for a couple more days.

closed 2024-12-16 by MRX8024

i3docdev, 2024-12-17:
See this was recently closed, but I am getting the same issue. Fails on internal error every time I run, even when running one axis at a time. Updated about 30 minutes ago.

[klippy (4).log](https://github.com/user-attachments/files/18159576/klippy.4.log)

i3docdev, 2024-12-17:
I believe that the same conditional change for beacon was also applied for [scanner], this may resolve in the same way.

LovesDonuts, 2024-12-17:
Just tested it. And it works fine.

MRX8024, 2024-12-17:
> See this was recently closed, but I am getting the same issue. Fails on internal error every time I run, even when running one axis at a time. Updated about 30 minutes ago.
> 
> [klippy (4).log](https://github.com/user-attachments/files/18159576/klippy.4.log)

Reboot the host or something similar.. If the error persists, create a new issue.

-Maxim