MRX8024/motors-sync #19: sync_motors not working?

KA009onPC, 2024-12-12:
Good afternoon! as of yesterday i started having issues with the sync motor function (after updating in my web interface)

before when motor sync would work i noticed X,Y,X1 and Y1 would turn off current temps when doing its function (you'll see in pictures) i notice now the only steppers that seem to do anything is X and Y, X1 and Y1 dont seem to do anything anyone (with motor sync) and when i run the SYNC_MOTORS to recalibrate and get my information to put into my .cfg this is what happens (please see attached) 
[image]
and it will just sit here and do nothing. 

it almost seems like sync_motors doesn't use X1 and Y1 anymore? I've done stepper buzz and i know X1 and Y1 works

if you have any question or want me to try things out please let me know!

Thanks!

KA009onPC, 2024-12-12:
so when i do the sync_motor command i will see x and y stepper current turning off and then it gets to the point where both stay off and this is all it shows (please see attached) and it never seems to finish motor sync and give me all the information i need?

Here is everything!
[klippy (4).lo
[image]
g](https://github.com/user-attachments/files/18117643/klippy.4.log)
[sync_stats.csv](https://github.com/user-attachments/files/18117646/sync_stats.csv)

okay that explains why i no longer see X,Y,X1 and Y1 turning on and off as well!

MRX8024, 2024-12-12:
You have fairly constant problems with the X axis. The script cannot decide which direction it should go, because the accelerometer shows different information each time, and simply completes the calibration for this axis, since no tolerance is set. And further actions depend on whether it worked stably for you before. How long have you been using this plugin? If there really haven’t been such cases before, I will prepare specific measurements that would help you understand what could have broken during the update in your situation. First, adjust the tolerance parameters, and see if the stepper can come to ~zero same as Y axis, before exceeding the maximum number of retries.
Otherwise, there may be hardware problems in the mechanics of the printer, for example, loose accelerometer mount, loose belts, etc. There are false readings when the stepper motor is in a “position between two sides” (when it can go in both directions and will take the same number of steps to arrive at the minimum point), in this position the readings are taken with a large spread. In your case, the magnitude is about 30k, which is not typical for this motor position. Therefore, I would also like to know in more detail about your printer, project, motors, belts, etc.
I really want to improve the logic of the program so that it can "solve" such cases and arrive at ~0. However, it is quite difficult to come up with some kind of ultimate solution. I do not have such situations on any of my printers, and I have nothing to test on)

KA009onPC, 2024-12-12:
I've been using the plug in for about 2 months now? 

I will gladly try to do anything you'd like, I'd love to help

I'll give a back story on my printer

Currently I am running a siboor trident AWD, I have manually (aligned) the steppers by turning steppers on and loosening the front pulleys so the motors are relaxed.

This printer also has Cartographer on it.

When I'm home I can also send you my printer.cfg if this will help? Maybe I have done something wrong

MRX8024, 2024-12-12:
I can see your configuration through the log file. 
Start with setup this -
> the parameters `retry_tolerance` and `retries`

Run one or more syncs, look at the result, attach the csv here again. Next we'll look.

KA009onPC, 2024-12-13:
i set the retries to 5, and i ran sync_motors 5 times as well and here is what i get

[sync_stats.csv](https://github.com/user-attachments/files/18132024/sync_stats.csv)

[image]

[image]

[image]

im not sure if this looks right, and also i still notice after all those runs it keeps the steppers turned off? (going off of the temp readings not showing)

also before i use to get information in the console on what to set the model too and the model_coeffs to set up in the .cfg

i hope this all helps!

KA009onPC, 2024-12-18:
Any update on this?

olibanez54, 2024-12-18:
> 
> im not sure if this looks right, and also i still notice after all those runs it keeps the steppers turned off? (going off of the temp readings not showing)
> 

I have the same question, after sync, X and Y steppers remains disabled with the las update, I noticed this because I defined the fan stepper drivers turns on/off following the X stepper Driver status, now the fan turns off a couple minutes after doing sync while mainsail dashboard is showing the steppers are active??

I mean I could change the fan logic to follow X1 driver, no big deal, but I feel this behavior is causing a little desync after syncing, cause I can hear a little “knock” sound when I home again.

MRX8024, 2024-12-19:
> Any update on this?

I don’t really understand what you want to hear from me, according to the latest data that you provided me, everything seems to be working for you.

> steppers remains disabled with the last update

Well, the motors really remain on after synchronization after 3cea525, I don’t see any particular problem with this, since with any subsequent movement they should turn on “at the same time,” but I’ll fix that.

KA009onPC, 2024-12-20:
does it seem to be working correct? okay i just wanted to make sure, the only thing i noticed is before it would give me information to set up the model too and the model_coeffs to set up in the .cfg, but now that doesnt show up, so should i delete the old values i put into my .cfg?

if everything else seems good thats awesome i just wanted to make sure

thank you so much for all your help

MRX8024, 2024-12-20:
> i noticed is before it would give me information to set up the model too and the model_coeffs to set up in the .cfg, but now that doesnt show up

I don't really understand what you mean. Nothing has changed there; after calibration, the algorithm outputs models and their coefficients to the klippy terminal, also generates a graph, and does nothing else. I plan to add auto-add to the configuration via `save_config` in the near future, as well as some other changes and improvements.

> so should i delete the old values i put into my .cfg?

yep