MRX8024/motors-sync #20: TypeError on raise for retry count

tristanmcpherson, 2024-12-16:
**Describe the bug**
A detailed description of what the bug is.

Too many retries error causes internal error on the command causing printer to shutdown instead of fail out of the print

**Expected behavior**
A description of what you expected to happen.

Raising error does causes print to fail and not an internal command exception

**Logs**
Attach `klippy.log`, screenshots to help explain your problem. 
If the problem is related to the motion system, attach `sync_stats.csv` from the directory where the repository was installed.

Too many retries
Internal error on command:"SYNC_MOTORS"
Traceback (most recent call last):
  File "/home/biqu/klipper/klippy/gcode.py", line 298, in _process_commands
    handler(gcmd)
  File "/home/biqu/klipper/klippy/gcode.py", line 207, in func
    return origfunc(self._get_extended_params(params))
  File "/home/biqu/klipper/klippy/extras/motors_sync.py", line 734, in cmd_SYNC_MOTORS
    self._run_sync()
  File "/home/biqu/klipper/klippy/extras/motors_sync.py", line 634, in _run_sync
    self.axes_level(max_ax, min_ax)
  File "/home/biqu/klipper/klippy/extras/motors_sync.py", line 573, in axes_level
    raise self._handle_state(m, 'error')
TypeError: exceptions must derive from BaseException
Transition to shutdown state: Internal error on command:"SYNC_MOTORS"

MRX8024, 2024-12-16:
Hi, this was fixed "yesterday", in a 1d6f6a4. You can update and make sure that it works. 
Thanks for the report!
-Maxim

tristanmcpherson, 2024-12-17:
I'm getting a different error now:

`Script running error
Traceback (most recent call last):
  File "/home/biqu/klipper/klippy/extras/delayed_gcode.py", line 41, in _gcode_timer_event
    self.gcode.run_script(self.timer_gcode.render())
  File "/home/biqu/klipper/klippy/gcode.py", line 321, in run_script
    self._process_commands(script.split("\n"), need_ack=False)
  File "/home/biqu/klipper/klippy/gcode.py", line 298, in _process_commands
    handler(gcmd)
  File "/home/biqu/klipper/klippy/gcode.py", line 389, in cmd_default
    raise gcmd.error(self.printer.get_state_message()[0])
gcode.CommandError: Internal error on command:"SYNC_MOTORS"
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown`

MRX8024, 2024-12-17:
Please attach full log