MRX8024/motors-sync #21: Error NaN in SYNC_MOTORS

jeffalanmcclain, 2024-12-17:
**Describe the bug**
Running SYNC_MOTORS after completing a SYNC_MOTORS_CALIBRATE klipper shuts down with can not evaluate NaN.

I have used:
model: exponential
model_coeffs: 6905.4832286225,0.0443039452,1045.9401821282

**Expected behavior**
Expected operation to complete successfully.
[image]

MRX8024, 2024-12-17:
Please attach klippy.log and sync_stats.csv from the directory where the repository was installed.
Your calibration looks pretty mediocre, tell us about your printer. The error is probably related to the third coefficient of the exponential function, which is 1045.94. When the magnitude goes below it, it gives NaN. The question remains why the magnitude was not able to drop below ~5000 during calibration.

jeffalanmcclain, 2024-12-17:
Hello,

sync_stats.csv is attached.  Klippy.log exceeds file size allowed to upload here...  I'm on a Siboor 350 AWD.  Motors have 32 microsteps, so that was what used (saw warning, however, I have 40 not 20 pulley).  When I ran 16 microsteps, it got to 500, actually, and better solve.

BR,
Jeff
[sync_stats.csv](https://github.com/user-attachments/files/18172435/sync_stats.csv)

MRX8024, 2024-12-19:
> (saw warning, however, I have 40 not 20 pulley)

Are you sure that you have a 40t pulley?, t is a teeth, a 20t pulley is equal to 40mm in a klipper configuration, since the tooth on a gt2 belt equal 2mm.

Well, no one is stopping you from testing whether 32 works in your case, but I would start with 16 and debug the system to see if everything works as it should. At the moment, if the calibration is performed for 16 microsteps, it is not recalculated to 32. But this also does not affect the worse, since the microsteps in sync process are calculated in half as many. I have plans in the future to change and improve the calibration, at the moment it is still quite "experimental"

I would be interested to look at your `sync_stats.csv` after some time, for example after 50-100 syncs. If you don't forget, bring it here sometime.

-Maxim

jeffalanmcclain, 2024-12-19:
Hello, Maxim.  Uh, you are correct.  I have a 40mm (I misunderstood the caution).  I will move back to sync at 16, even though my motor control uses 32 microsteps.  I will also stop back by here and provide the sync_stats.csv when that is done.

In any event, you should likely put in try-except clauses anywhere NaN values could occur and handle accordingly, even if they stem from ill-advised operation of the calibration routines.

Thank you and Best Regards,
Jeff