MRX8024/motors-sync #24: motors_sync: Not support 1' count of motors

ConstantinMA, 2025-01-01:
After motors-sync inslallation: Klipper reports: ERROR

motors_sync: Not support 1' count of motors
[klippy.log](https://github.com/user-attachments/files/18286808/klippy.log)

MRX8024, 2025-01-02:
Hello, what goals do you want to achieve by installing a sync script, having 1 motor on the belt?

ConstantinMA, 2025-01-02:
I apologize, apparently I should have explicitly stated: it's a regular corexy printer.

MRX8024, 2025-01-02:
Script synchronizes two motors, which the klipper controls separately, two drivers. In your printer configuration (according to klippy.log), only one motor (e.g. `stepper_x`) is registered for each axis. That's why it's not clear to me what you're doing and why.

ConstantinMA, 2025-01-02:
Yes, in the case of CoreXY kinematics the stepper_x (stepper_y) designation is arbitrary, to move along the x-axis (or y-axis) both motors must operate synchronously: 
https://en.wikipedia.org/wiki/CoreXY
https://www.klipper3d.org/Config_Reference.html?h=corexy#corexy-kinematics
However, I think this is a common designation:
https://github.com/Klipper3d/klipper/blob/master/config/example-corexy.cfg
And I wanted to investigate whether synchronizing the motors would help reduce the difference between the vibration energy when moving along the x (y) axis and diagonally, as in this graph: 
https://github.com/Frix-x/klippain-shaketune/blob/main/docs/macros/create_vibrations_profile.md#polar-angle-energy-profile 

Please forgive me if I'm misunderstanding.

MRX8024, 2025-01-02:
You misunderstood. This script was created to synchronize ***two motors on one belt***, for example: stepper_x with stepper_x1.
It does not in any way synchronize the corexy kinematics itself, and I don’t really understand what you want to synchronize in it, it does not lend itself to this.

ConstantinMA, 2025-01-02:
Thanks, now I realize that I was misinterpreting the applicability of the script to corexy kinematics itself.
I was hoping to improve the interaction between stepper_x and stepper_y motors, since in CoreXY kinematics they almost always work together, albeit at different speeds. Apparently this is not easy to do.
Again, I apologize and thank you for the clarification.

closed 2025-01-02 by ConstantinMA