MRX8024/motors-sync #25: [Bug] SYNC_MOTORS causes kalico to crash

sbtoonz, 2025-01-04:
**Describe the bug**
When running sync_motors the motor sync beyond my desired threshold and eventually crashes klipper

**Expected behavior**
For sync to stop once threshold met

**Logs**
Attach `klippy.log`, screenshots to help explain your problem. 
[klippy (20).log](https://github.com/user-attachments/files/18308257/klippy.20.log)

If the problem is related to the motion system, attach `sync_stats.csv` from the directory where the repository was installed.
[sync_stats.csv](https://github.com/user-attachments/files/18308258/sync_stats.csv)

Screenshots:
[image]
[image]

MRX8024, 2025-01-05:
This problem is now ~opened in #21, and I'm thinking about a solution to such cases.
It would be nice if you could include the calibration graph from which you derived these steps models.

Thanks,
-Maxim

sbtoonz, 2025-01-05:
I do believe they get overwritten after a subsequent run?

MRX8024, 2025-01-05:
No, the graphs in the name sign the time, so new ones are generated

sbtoonz, 2025-01-05:
Ok I'll drop them in this issue
For now I returned a gcode error that there was NaN instead of hard
crashing

MRX8024, 2025-01-05:
> For now I returned a gcode error that there was NaN instead of hard crashing

You are generally right, but I am thinking about solving the problem with linefit so that such cases, in principle, never arise. Maybe I'll add a check to the calibrator, but I really don't want to add it to the main script. Perhaps I will completely abandon the exponential function, in view of the already three open issues related to it.

> **Expected behavior**
> For sync to stop once threshold met

In fact, this would be logical, but at the moment it is simply not possible. I believe that the motors should synchronize until the end, and tolerance was designed to eliminate cases of false accelerometer measurements (quite a lot of problems), and premature completion of calibration.
At the moment, accelerometer noise is not calculated very correctly, at least in the median filter. For example, when the fans are turned on, the minimum magnitude threshold may increase, which will disrupt your proposal. Also, a slight change in the printer in space can cause deviation. Of course it shouldn't be this way, but I haven't arrived at what it should look like. If you have specific suggestions, I can listen to them.

sbtoonz, 2025-01-05:
[image]
[image]

sbtoonz, 2025-01-05:
```
result = m.model_solve()
if np.isnan(result):
    raise self.gcode.error(f"Model solve returned NaN for axis {m.name}")
```

between these 2x lines

https://github.com/MRX8024/motors-sync/blob/main/motors_sync.py#L850-L851

is what i did to avoid klipper just dying

AstronIsTaken, 2025-01-06:
sync_motors started crashing for me as well after update and recalibration

MRX8024, 2025-01-06:
> sync_motors started crashing for me as well after update and recalibration

Also provide your klippy.log and a calibration graph if possible.

AstronIsTaken, 2025-01-06:
[
[klippy_motors_sync.log](https://github.com/user-attachments/files/18323912/klippy_motors_sync.log)
[image]
[sync_stats.csv](https://github.com/user-attachments/files/18323913/sync_stats.csv)
](url)

Looks like recalibration changed model from exponential to power and sync_motors doesn't fail any more (at least for now). 

Previous model (which used to fail):
```
[motors_sync]
steps_model_x = exponential,
	20.917766234773918,
	0.4918042586607617,
	1649.1713178947184
steps_model_y = exponential,
	20.917766234773918,
	0.4918042586607617,
	1649.1713178947184
```

New model (after 2nd calibration) that works so far:

```
[motors_sync]
steps_model_x = power,
 	  1340.7646633231607,
 	  1.2724183376736313
steps_model_y = power,
 	  1340.7646633231607,
 	  1.2724183376736313
```

AstronIsTaken, 2025-01-06:
[image]
Graph for first calibration that resulted in exponential model. Is it supposed to change between calibrations? I think I slightly tightened one of the belts between calibrations

MRX8024, 2025-01-06:
> Is it supposed to change between calibrations?

Ideally, it shouldn’t, some measurement error is allowed, but I have never seen graphs like your first one (presented in this post). His dependence looks strange, and the all functions did not fit well. The second one with the power function looks much better, it is also closer to a linear relationship.

sbtoonz, 2025-01-07:
I'll pull your changes down and test

MRX8024, 2025-01-10:
I think we will close this issue, since the problem described is actually solved, even though it was indirectly caused by the calibrator. Soon I will open a new where I will continue the topic of improving the calibrator. I'll mention you in it so you can test it if you want.

closed 2025-01-16 by sbtoonz