MRX8024/motors-sync #26: sync_motors macro

axg20202, 2025-02-18:
It looks as though the sync_motors command does not appear as a macro as it isn't really a macro. 

Is there a way to link the command to a macro so that it can be assigned as a macro button on the GUI? It would make it a lot easier to use outside of running gcode as there would be no need to type the command in the terminal.

MRX8024, 2025-02-18:
Hi, yes you can:
```
[gcode_macro SYNC_MOTORS]
rename_existing: SYNC_MOTORS_BASE
gcode:
    SYNC_MOTORS_BASE {rawparams}
```

-Maxim

axg20202, 2025-02-18:
Thanks. Strangely, the macro still doesn't appear in the list of available macros in Mainsail.

MRX8024, 2025-02-18:
It should have appeared because its just a standart macro. Try restarting your browser, mainsail, etc.

axg20202, 2025-02-18:
Definitely doesn't show up. If I add a test macro it shows up instantly, so not a browser refresh issue.

axg20202, 2025-02-18:
Wierdly, it doesn't like to be renamed. If it's named as [gcode_macro MOTORS_SYNC], get error:
Existing command 'MOTORS_SYNC' not found in gcode_macro rename

MRX8024, 2025-02-18:
In this case you shouldn't use `rename_existing`, the error seems to be related to this

axg20202, 2025-02-18:
I'm guessing because the macro doesn't really exist to be renamed

drb1981, 2026-04-18:
This is an old thread, but for anyone coming here trying to figure this out like me, klipper simply doesn't like using the same command even if it has something on the end.  Need to break it up with a different word or something.  This works:
```
[gcode_macro SYNC_ALL_MOTORS]
gcode:
    SYNC_MOTORS
```