MRX8024/motors-sync #28: 'NoneType' object has no attribute 'get_steppers'

AlexTheGreat1971, 2025-04-18:
SYNC_MOTORS works and completes correctly. However, if you run SYNC_MOTORS_CALIBRATE, Klipper immediately crashes with an error: 'NoneType' object has no attribute 'get_steppers'. Stats and part of klippy.log are attached.

Voron Trident 350 AWD
Armbian 25.2.3 bullseye

Thanks for your help

[sync_stats.csv](https://github.com/user-attachments/files/19815034/sync_stats.csv)

[klippy.log](https://github.com/user-attachments/files/19815035/klippy.log)

MRX8024, 2025-04-18:
Hi, thanks for the report. Please provide the command you ran with all the parameters. I think I heard about a similar error recently.

AlexTheGreat1971, 2025-04-18:
In printer.cfg:
[motors_sync]
axes: x,y
#    Оси на которых будет производиться калибровка.
accel_chip: adxl345

Command just is: SYNC_MOTORS_CALIBRATE AXIS=X

AlexTheGreat1971, 2025-04-18:
Weird, it’s working now (SYNC_MOTORS_CALIBRATE AXIS=X)... 
Klipper had been crashing for two days straight before this. 
Never mind—false alarm.
Thanks for the nice product.