MRX8024/motors-sync #31: Support for generic_cartesian Kinematics for like 6WD Hybrid CoreXY

N13Dp, 2025-06-08:
Hi,
I’d like to suggest adding support for generic_cartesian kinematics to motors-sync. This would allow more advanced setups (like multi-motor systems using [carriage] and [extra_carriage], e.g. 6WD printers) to use this tool without needing to switch to classic cartesian.

It would also be helpful to have an optional config like:

ini
Kopieren
Bearbeiten
manual_axes_override: x,y,z
This way users can manually define which axes to sync, regardless of the detected kinematics.
Both changes would make motors-sync more flexible without breaking existing setups.

Thanks for considering!

MRX8024, 2025-06-16:
Hi, I agree that this would be a really cool update, and this project is currently moving a little towards adding new kinematics. I haven't read generic_cartesian merged code yet, but I read the discussion threads when it was being created and generally understand the idea. So, at the moment I just can’t imagine how this could be added to motors-sync. In case of 6WD Hybrid CoreXY there should be several stages of syncs. If I understand correctly it is 4wd corexy and 2wd hybrid. So you should first sync corexy motors, and then sync hybrid motors to these motors. This can be written in software, and even one person did it. However, generic_cartesian kinematics assumes that you can describe anything in it, and motors-sync must somehow understand what kind of kinematics you have. In automatic mode, I have a hard time imagining how this can be achieved. (thoughts) Most likely, the user will have to describe in the configuration which carriages he needs to sync with which, without violating the order or something like that. If you have any thoughts on this, I'd be happy to hear them. 

In any case, my current plans are to fix some issues with the current software, and write a base for easy addition of new kinematics.

-Maxim

MRX8024, 2025-12-06:
No, motor_sync still doesn't support any kinematics other than corexy and cartesian for bedslingers. Although I plan to finally complete and release the abstraction for various kinematics this week, I still have no idea how to describe (for motor_sync) such complex kinematics in the configfile. I'd suggest you contact me (contacts is at the bottom of the doc) and attach your config file or klippy log so I can fully understand the implementation and description of this kinematics. And then if I start implementing it, it will be much more convenient to communicate with you in pm to test the code.