MRX8024/motors-sync #33: Microsteps calculation returned NaN for X axis

DocFoFo, 2025-06-28:
**Describe the bug**
Any function causes this error, 
Microsteps calculation returned NaN for X axis. Everything is updated. The axis being reported NaN changes but the error reads the same. I removed motor_sync and reinstalled, no luck. There are not custom variables in the CFG. This only started after an update. 

**Expected behavior**
Sync Motors, Calibration, etc., to complete and not error out. 

[klippy.log](https://github.com/user-attachments/files/20962154/klippy.log)

[sync_stats.csv](https://github.com/user-attachments/files/20962160/sync_stats.csv)

**Logs**
Attach `klippy.log`, screenshots to help explain your problem. 
If the problem is related to the motion system, attach `sync_stats.csv` from the directory where the repository was installed.

DocFoFo, 2025-07-06:
I figured out the NaN issue; I hadn't removed the calibration data from the saved data at the bottom of my CFG file. Now, for the graph issues, I can't figure it out. I've gone through the gantry twice. No obvious issues. No loose parts or binding anywhere. Good times.

MRX8024, 2025-07-06:
You can attach updated data (logs, graphs) and I will take a look at them. Tell, show your printer, etc. More information, even if you think it is uninformative, may help us understand the problem.