MRX8024/motors-sync #35: 3 motors per belt loop (coreXY)

riley-github, 2025-08-01:
I am designing a corexy motion system with 3 nema17s per belt loop. This is a larger ish printer so I don’t want the belts path to be too long without a motor so 3 motors per path makes the most sense. I know someone asked about 6WD with a hybrid motion system but would awd sync work for 3 motors per loop?

 I love this project and think it is one of the most important klipper add ons for high speed printing!

[image]

MRX8024, 2025-08-06:
Hi, there is no support for this at the moment, and it is not that easy to implement. I do not quite understand your goal, when people use awd, they divide the belt section with two motors so that the belt of the "moving kinematics", that is, the "useful length of the belt", is as short as possible. In general, it doesn't matter to us at all how long the belt is in the unused part of the kinematics, since we have a strictly limited belt length between two motors on the used side, from which the loads act. Therefore, I don't understand what another (3) motor will do, which breaks up the belt sections to which (practically) no loads come.

riley-github, 2025-08-06:
Hi I would argue that the 3rd motor is shortening the useful length of the belt even further. I guess I’m not understanding your point. What do you mean when you say unused part of the kinematics? I need the extra torque from the third motor to help drive a heavy gantry. I would be interested in talking about this more and learning your perspective

MRX8024, 2025-08-06:
There are two sections of the belt in the awd. The first one goes through the joints, beam, toolhead, and regularly receives loads. The second one goes on the other side of the motors, and goes around the portal to loop the path of the entire belt. So, I think that the second section of the belt practically does not participate in the loads of the first section of the belt, because the two sections of the belt are rigidly separated by the motors, and the loads on the two different sections of the belt do not interact with each other. So, you are going to build a third motor into a section of the belt, which in my opinion (practically) does not take on any loads, but exists only to loop the entire belt. A third motor could indeed increase peak torque, but I don't think it would contribute to the rigidity of the entire system.

Are you building this printer for yourself? When you program things like this, it's hard to get it right the first time without testing it on the printer next to you. So even if I were to start developing this, I'd need someone to test it, and probably more than once.

riley-github, 2025-08-06:
https://youtu.be/NTVJZ5p2U2k?si=u_nodHsT_qmF4UBH

Yes I’ve been developing this machine for almost a year. I’ve attached a video of the most recent iteration. 

So if I am understanding you correctly the motors that pull directly on toolhead affect rigidity. So technically in the picture of the belt paths I attached above by adding the extra motor on the blue belt path in the bottom left I am making the motor in the upper left much less needed?

If I am to go through with adding the new motors it would be for the torque gain. I currently have a mgn9 rail with no preload so I am having problems with toolhead nodding. I have an mgn12 rail with much less play but adding this will require more torque. My two options are I could either just get a new rail or I could upgrade motors and get new rail for more speed. My goal is to get under 3 mins for benchy on a larger machine. (Silly goal I know but if I can get this large prints will be fast)

MRX8024, 2025-08-10:
So, I think that the closer the motor is and the more it reduces the length of the belt that goes through the beam and toolhead, the better, and what the third motor will give I am not sure, and this needs to be tested. In your photo above, both belts go forward of the portal, and this is no longer corexy kinematics, and I generally do not understand how this kinematics should work.

Support for three-motor sync may be added, but not in the near future. Or you can implement it yourself, the project is open source. I already have a lot of other plans for this project, and there are some priorities.