MRX8024/motors-sync #37: update_manager motors-sync klipper error

DBRUNOEnder, 2025-09-08:
Hello,
my printer, after start, shows problem with future update:
Future update not avaible. 'requirements: wiki/requirements.txt

Can You please help? :)

MRX8024, 2025-09-08:
I don't know why moonraker writes this, maybe it is directly related to your 

`Unsafe Shutdown Count: 158`

Try running the `git fetch --all && git reset --hard origin/main && git clean -fd` in the script directory and restart moonraker/klipper

DBRUNOEnder, 2025-09-09:
[image]

DBRUNOEnder, 2025-09-09:
Maybe i have probleme here

DBRUNOEnder, 2025-09-09:
[image]

JNSmith1717, 2025-09-13:
Had the same issue, ended up deleting the motors_sync folder and removed part from moonraker.conf, then reinstalled, issue gone